HaarigerHarald/omxiv #2: libjpeg8-dev unavailable but required?

DerRheingold, 2017-11-04:
Hi,
awesome project! Just what I was looking for :)
However, when trying to install libjpeg8-dev I get this message on my Raspi running Kodi/OSMC: 

"Package libjpeg8-dev is not available, but is referred to by another package. This may mean that the package is missing, has been obsoleted, or is only available from another source However the following packages replace it: libjpeg62-turbo-dev
E: Package 'libjpeg8-dev' has no installation candidate"

If I install the version libjpeg62-turbo-dev and try to run "make ilclient" I get further error messages:
"mkdir -p libs
cp -ru /opt/vc/src/hello_pi/libs/ilclient libs cp: cannot stat '/opt/vc/src/hello_pi/libs/ilclient': No such file or directory
Makefile:16: recipe for target 'ilclient' failed
make: * [ilclient] Error 1"

Any idea how to fix/circumvent this?﻿

Cheers :)

nagualcode, 2017-11-05:
probably your system does not have the raspberry fimware files and libs:
/opt/vc/src/hello_pi/
You my copy it from the official raspberry firmware repo in github,
I suggest you to use raspbian image if you want to play with programming yourself.

DerRheingold, 2017-11-05:
Hi,

thanks for the quick reply :)
In fact I don't intend to program on my raspberry "just" make some small modifications, like getting it to run as chromecast ;)

So I copied the /opt/vc/src/hello_pi/ from the raspberry firmware to my raspberry and made some progress. Now I get another error message
" fatal error: interface/vcos/vcos.h: No such file or directory
 #include "interface/vcos/vcos.h" "
However in the firmware vcos is located in "opt\vc\include\interface\vcos" having copied the files to that location won't fix the problem. 

Any ideas?

HaarigerHarald, 2017-11-05:
I'd try to update the firmware with rpi-update. Afaik this overwrites the opt folder with the needed files and file structure, in case that's the problem. 

Moreover I'm pretty sure you can also use other versions of libjpeg.

Ruffio, 2018-08-26:
@DerRheingold can this issue be closed?

DerRheingold, 2019-01-08:
Sorry, I think you can close it. I found another way to solve my initial problem... but truth be told, I don't remember it anymore ^^

Ruffio, 2019-01-27:
@DerRheingold can you close this?

closed 2019-01-28 by DerRheingold