HaarigerHarald/omxiv #4: Unable to cast after setup- No errors

dilipprasad, 2018-01-18:
Hi, 
Even after setting up, i am not receiving any errors. Kindly help here.

How to start manually? Phone says the video is being casted from my phone, but unable to see in the screen.

Ruffio, 2018-08-26:
Did you find out how to resolve this?

Ruffio, 2019-01-27:
@dilipprasad can this issue be closed?

Schattenruf, 2019-11-25:
Hi,

have the same issue here.
My phone says "Streaming with active wakelock" but on my pi nothing happens. IP & User are correct and I have downloaded omxplayer and followed the instructions here.

(I was on a non gui raspbian before. Now I downloaded the gui because I thought this was the problem).

nagualcode, 2019-11-25:
can you point me the link to where it says omxiv is a casting/streaming solution?

Schattenruf, 2019-11-26:
Well it is if you use it with Raspicast. I have tested it with a new Clean
Installation of Raspbian Buster and that works just fine.

nagualcode, 2019-11-26:
Well, "Raspicast" is the casting/streaming software.
Omxiv may be just a dependency, Raspicast needs to work. But in fact, Omxiv is just a photo (jpg/png files) viewer for the raspberry, not even a video player.
So this casting issue is not related to Omxiv at all.

Ruffio, 2020-08-26:
Please close this.

closed 2020-09-20 by HaarigerHarald