HaarigerHarald/omxiv #5: Blend animation - unable to get it working

colloqi, 2018-03-22:
Thanks for the excellent code! I am unable to get the animation working on Pi 3. Is there a configuration I am missing in config.txt? 
command - omxiv -t 5 -T blend --duration 1200 *.jpg
everything works except alpha is always 255 (irrespective of any value given under testing)

Thanks once again.

colloqi, 2018-03-27:
I could get it working after changing the resize colorformat to OMX_COLOR_FormatYUV420PackedPlanar 

Thanks

closed 2018-03-27 by colloqi

nagualcode, 2018-07-24:
where did you made that change?

colloqi, 2018-07-25:
[image]

nagualcode, 2018-07-25:
Thank you.
It worked.
I sent a pull request here.

HaarigerHarald, 2018-07-26:
Hi,

the problem with this workaround is that you will lose pixel data for everything other than normal jpegs. When I wrote the part with the blend transition the argb alpha channel was ignored but display config alpha was respected, now it's the other way around, which imho isn't correct either. A proper fix therefor would require some firmware change. The best workaround I guess would be to render in the same format as the decoded image

nagualcode, 2018-07-29:
How can I make omxiv quits after one iteiration over the imagefiles in dir?

commit 871981b4 (2018-11-17): Fix blend transition by using new alpha flag
Fixes #5