HaarigerHarald/omxiv #11: soft_image.c:33:21: fatal error: jpeglib.h: No such file or directory  #include <jpeglib.h>

cemox35, 2018-10-13:
cc -DVERSION=\"18e047dfb6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c soft_image.c -o soft_image.o -Wno-deprecated-declarations
soft_image.c:33:21: fatal error: jpeglib.h: No such file or directory
 #include <jpeglib.h>
                     ^
compilation terminated.
Makefile.include:11: recipe for target 'soft_image.o' failed
make: *** [soft_image.o] Error 1

How can I fix this error?

HaarigerHarald, 2018-10-14:
Did you install the development package for libjpeg?

cemox35, 2018-10-21:
Yes, However, I didn't understand why cannot install development packages. If I run install command returns about libjpeg64-turbo package.