HaarigerHarald/omxiv #12: Cannot compile

harddevelop, 2018-11-25:
When I try to launch the first make I have this error:

pi@raspberrypi:/opt/omxiv $ make -j4                                         echo -n "static void printUsage(){printf(\"" > help.h                        sed -n -e '/\#\# Synopsis/,/\#\# / p' README.md | sed -e '1d;$d' | sed ':a;N;$!ba;s/\n/\\n/g' | tr '\n' ' ' >> help.h                                     echo "\\\\n\");}" >> help.h                                                  cc -DVERSION=\"2f27c1d4a6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c omxiv.c -o omxiv.o -Wno-deprecated-declarations                                                                       cc -DVERSION=\"2f27c1d4a6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c omx_image.c -o omx_image.o -Wno-deprecated-declarations
cc -DVERSION=\"2f27c1d4a6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c omx_render.c -o omx_render.o -Wno-deprecated-declarations
cc -DVERSION=\"2f27c1d4a6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c soft_image.c -o soft_image.o -Wno-deprecated-declarations
omx_render.c: In function ‘setOmxDisplayConfig’:
omx_render.c:309:21: error: ‘OMX_DISPLAY_ALPHA_FLAGS_MIX’ undeclared (first use in this function)
   dispConfRT.alpha|=OMX_DISPLAY_ALPHA_FLAGS_MIX;
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
omx_render.c:309:21: note: each undeclared identifier is reported only once for each function it appears in
Makefile.include:11: fallo en las instrucciones para el objetivo 'omx_render.o'
make: *** [omx_render.o] Error 1
make: *** Se espera a que terminen otras tareas....

HaarigerHarald, 2018-11-26:
The alpha flag was introduced recently, so you need to update your firmware. I think it's already in the main repo, therefor apt-get upgrade should do it.

harddevelop, 2018-11-27:
Incredible, you're right. 

After apt-get upgrade OMX_DISPLAY_ALPHA_FLAGS_MIX appears and compile fine. 

Thank you so much

closed 2018-11-27 by harddevelop