HaarigerHarald/omxiv #16: Videos not playing

herna54, 2019-02-03:
I'm using a pi 3 and casting images works fine but casting videos leads to a Black screen and then closes the screen. Do I need to install something else?

Ruffio, 2019-02-03:
You do know this is an image viewer, right?

[image]

In order to play video you need a video player...

herna54, 2019-02-03:
Yes I do have it. Code I did was
'sudo apt-get install omxplayer'

matteoraggi, 2019-05-28:
yes, to play videos from cli, I suggest you this app wth a similar name: https://github.com/popcornmix/omxplayer

FerretDave, 2019-06-14:
The instructions for 'raspicast' refer to using both OMXiv and OMXplayer, easy mistake to make

Ruffio, 2020-08-26:
Please close this issue

laddu38732, 2020-09-20:
The answer to all questions

$ sudo apt-get install libjpeg8-dev libpng12-dev
$ git clone https://github.com/HaarigerHarald/omxiv
$ cd omxiv
$ make ilclient
$ make -j4
$ sudo make install

And enjoy|||||||

closed 2020-09-20 by HaarigerHarald

kkboy20, 2023-01-30:
Raspicast app does not work for youtube anymore