HaarigerHarald/omxiv #17: Error while executing "make"

E-Mans-Application, 2019-02-09:
Hello,

I'm trying to install omxiv on my Raspberry Pi 3.
I cloned with git then I ran `make ilclient` successfully.
https://pastebin.com/n0yB6iCG

But when I run the command "make", I got the error:
```
Makefile.include:11: recipe for target 'omx_render.o' failed
make: *** [omx_render.o] Error 1
```
Running with sudo does not solve the issue.
https://pastebin.com/nBH9MHVx
https://pastebin.com/RDdpVFm5

Favrito, 2019-02-10:
I have this same problem, please someone help!

HaarigerHarald, 2019-02-10:
Duplicate of #12 and #14

E-Mans-Application, 2019-02-10:
I have already run "apt-get upgrade" (took almost an hour)
I'm trying another update then I'll tell you if this is working fine.

nagualcode, 2019-02-10:
"Building
You will **need the development header for libjpeg and libpng (Debian: libjpeg8-dev, libpng12-dev).** Then build it on the Pi with:

**make ilclient**
**make**
And install with:

sudo make install"

E-Mans-Application, 2019-02-10:
I had already installed libjpeg and libpng.
After running : `sudo apt-get update && sudo apt-get -y upgrade && sudo apt-get -y dist-upgrade && sudo rpi-update && sudo reboot`, the installation of omxiv worked. I think "rpi-update" made the difference.

jackdaniels92, 2019-02-21:
> I had already installed libjpeg and libpng.
> After running : `sudo apt-get update && sudo apt-get -y upgrade && sudo apt-get -y dist-upgrade && sudo rpi-update && sudo reboot`, the installation of omxiv worked. I think "rpi-update" made the difference.

Thats correct. "rpm-update" made the difference. Thanks a lot.

bitstreamreport, 2019-02-21:
It is important to notice that this is a raspbian distro issue, not omxiv.
Raspbian seems to have a conservative way to update the firmware libs, and this "rpi-update" tool, forces the update to the lastest version. With other distros, the libs can be already updated.

rebroad, 2019-09-11:
is it possible to get this working without needing to do an rpi-update?

nagualcode, 2019-09-11:
> is it possible to get this working without needing to do an rpi-update?

It is important to notice that this is a raspbian distro issue, not omxiv.
Raspbian seems to have a conservative way to update the firmware libs, and this "rpi-update" tool, forces the update to the lastest version. With other distros, the libs can be already updated.

Ruffio, 2020-08-26:
Can this issue be closed?

closed 2020-09-20 by HaarigerHarald