HaarigerHarald/omxiv #18: Unable to locate

bastienclement-zz, 2019-04-12:
Hi there,

First of all thank you very much for developing this! I am now attempting to turn my raspberry pi into a chromecast :)

I am running into an issue when running sudo apt-get install libjpeg8-dev libpng12-dev, these are what the command line returns:

E: Unable to locate package libjpeg8-dev
E: Package 'libpng12-dev' has no installation candidate

Is there a workaround?

Cheers,
Bastien

MatthewBardijn, 2019-05-24:
I have the same problem and libjpeg62-turbo doesn't work.

Ruffio, 2020-08-26:
Is this still an issue or can be closed?

Ruffio, 2020-10-07:
This issue should be closed due to inactivity

SwordFish7249, 2021-07-17:
Hey did you solve this problem?Because when I was try to install mysql-workbench on my kali it gives me that error:
"Depends: libjpeg8 (>= 8c) but it is not installable
E: Unable to correct problems, you have held broken packages."
Then when I was run apt-get install libjpeg8 or apt-get install -f libjpeg8 on my terminal it gives me that error:
"Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libjpeg8"
So do you know how can I solve this issue?

ffloress, 2021-07-17:
libjpeg8-dev