HaarigerHarald/omxiv #23: Long transitions are going jerky

matteoraggi, 2019-05-28:
I tested this code and it is wonderfull! I am using las tsable raspbian on rapsberry pi 3 model b+.
With transitions of half seconds it is so great! If instead I do transitions of 10 second then it go jerky. It not happen with pi3d instead. I hope that you can get it better also to make also long slides to works perfect as we all would love. have great time, thanks for omxiv!

kaushiksv, 2019-08-15:
Same here with pi4b. I am anyway happy because this tool is way better than gthumb.

Ruffio, 2020-08-26:
Have to got some images and command line that you use to execute omxiv that reproduces the issue?

Ruffio, 2020-10-06:
This issue should be closed due to inactivity