HaarigerHarald/omxiv #24: Impossible to exit

matteoraggi, 2019-05-28:
I am using raspbian stretch last stable with raspberry pi 3 model b+.
With both Terminal and CTRL+ALT+F1, I can't exit using Q or ESC even CTRL+ALT+CANC is not working to exit.

FerretDave, 2019-06-14:
Likewise, using raspicast on android to cast images to rasbian, until I specifically tell raspicast to *stop* casting, the image remains and cannot be bypassed (even shutting down the tablet doesnt stop it!).

A command line option to 'stop showing the image after x seconds' would be handy..., as well as a way directly from the Pi to quit.

kaushiksv, 2019-08-15:
Had the same problem. I had to pkill via ssh.

Ruffio, 2020-08-26:
Do you have some test images that omxiv can't exit from?
What does raspicast has to do with omxiv?

Ruffio, 2020-09-20:
This issue has awating feedback from originator for 25 days. It should be closed due to stalling IMHO...

Ruffio, 2020-10-06:
This issue should be closed due to inactivity