HaarigerHarald/omxiv #28: Problem with some jpeg - decoder returned 0x10

iszczyg, 2019-12-22:
Hi,
I've just installed omxiv on RP4. It works perfectly, it is very fast and light. I really like it. But I have noticed that it can't show some jpgs and reports "decoder returned 0x10". These jpgs come from Samsung Galaxy S9. If I transcode the jpeg with jpegtran omxiv is able to show the picture. How to correct this problem?

Best regards
Irek, Poland

Ruffio, 2020-08-26:
Can you provide some images that fails as described above? Without concrete test data this will be impossible to persue...

Ruffio, 2020-10-02:
@HaarigerHarald This issue should be closed due to lack of response from originator in two months

mLojowski, 2020-12-28:
Hi.
I would like to reopen the issue as I've seen the same behavior with omxiv.  Was this solve before or we can reopen the issue?
I've got some pictures that return "decoder returned 0x10" notyfication, and I'm able to share some image with it. Regards,
Marcin

FlapTrap99, 2021-12-01:
Hi, I'm also having this issue. Two jpg's of the same scene, one decodes OK but the other returns 0x10.

[image]
[image]
.