HaarigerHarald/omxiv #30: Issue with "make" commands on DietPi OS

CartoDB2018, 2020-02-19:
Hello,

I am currently attempting to install Raspicast and use omxiv among others.

I've been following the below URL guide and reached step 4:

https://thepi.io/how-to-use-your-raspberry-pi-as-a-chromecast-alternative/

When entering "$make ilclient" under the 'omxiv" directory I get the following response:
INPUT:
dietpi@DietPi:~/omxiv$ make ilclient

OUTPUT:
mkdir -p libs
cp -ru /opt/vc/src/hello_pi/libs/ilclient libs
cp: cannot stat '/opt/vc/src/hello_pi/libs/ilclient': No such file or directory
make: *** [Makefile:25: ilclient] Error 1

Further, when trying the "make" command I get:

INPUT:
dietpi@DietPi:~/omxiv$ make

OUTPUT:

cc -DVERSION=\"2f27c1d4a6\" -DLCURL_NAME=\"libcurl.so.4\" -O3 -fdata-sections -ffunction-sections -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -pipe -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I./libnsbmp -I./libnsgif -I./libs/ilclient -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -g -c omxiv.c -o omxiv.o -Wno-deprecated-declarations
In file included from omxiv.c:14:
omx_render.h:32:10: fatal error: ilclient.h: No such file or directory
 #include "ilclient.h"
          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile.include:12: omxiv.o] Error 1

Lastly, I attempt the third command of the guide and get:

INPUT:
dietpi@DietPi:~/omxiv$ sudo make install

OUTPUT:
install omxiv.bin /usr/bin/omxiv
install: cannot stat 'omxiv.bin': No such file or directory
make: *** [Makefile:29: install] Error 1

At this point I am unsure as to the issue. I've followed all steps mentioned in the guide, and even took care to be in the Omvix directory prior to installing the libraries etc.

I looked at the issue posted here:

https://github.com/HaarigerHarald/omxiv/issues/17

It does not help as I cannot use the commands in question. Note, I am running DietPi.

I appreciate any help here and can provide more info.

Thanks!

nagualcode, 2020-02-20:
This is not related to Omxiv.
Your "Diet Pi OS" does not have those libs installed:
/opt/vc/src/hello_pi/libs/

CartoDB2018, 2020-02-20:
> This is not related to Omxiv.
> Your "Diet Pi OS" does not have those libs installed:
> /opt/vc/src/hello_pi/libs/

Thank you for your comment. Do you think using Raspbian would correct this problem?

nagualcode, 2020-02-22:
Yes, those libs comes pre-installed with raspbian.
Also you should be able to install on your distro:
https://github.com/raspberrypi/userland

Ruffio, 2020-08-26:
Has this issue been resolved and can be closed?

BurritoTown, 2020-08-26:
Hello,

This has been solved. Thanks.

Ruffio, 2020-08-26:
Then please close it ;-)

laddu38732, 2020-09-20:
The answer to all questions

$ sudo apt-get install libjpeg8-dev libpng12-dev
$ git clone https://github.com/HaarigerHarald/omxiv
$ cd omxiv
$ make ilclient
$ make -j4
$ sudo make install

And enjoy|||||||

Ruffio, 2020-09-20:
@BurritoTown would you close this?

closed 2020-09-20 by HaarigerHarald

Shabinder, 2020-10-20:
Same error on Ubuntu Mate , can we get the missing libraries somehow to resolve this issue:
cp: cannot stat '/opt/vc/src/hello_pi/libs/ilclient': No such file or directory