HaarigerHarald/omxiv #33: Error while running make

isThatHim, 2020-09-25:
Hello,
i cloned the git repo, ran 
`make ilclient` successfully, and then tried to run `make`.

Sadly the command fails with 

> sudo make -j4
> cc -o omxiv.bin omxiv.o omx_image.o omx_render.o soft_image.o ./libnsbmp/libnsbmp.o ./libnsgif/libnsgif.o -lilclient -ljpeg -lpng -lrt -ldl -Wl,--gc-sections -s -L/opt/vc/lib/ -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
> /usr/bin/ld: omx_render.o:/home/alarm/omxiv/omx_render.h:71: multiple definition of `OMX_RENDER_TRANSITION'; omxiv.o:/home/alarm/omxiv/omx_render.h:71: first defined here
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:13: omxiv.bin] Error 1

I am running
> Linux pi 5.4.65-1-ARCH #1 SMP PREEMPT Tue Sep 15 16:21:17 UTC 2020 armv7l GNU/Linux

and i previously installed [raspberrypi/userland](https://github.com/raspberrypi/userland) to run`make ilclient` successfully.

I also installed libjpeg and libpng from the Arch-Repos.
I updated and restarted my system.

Does someone know what i did wrong?

Thanks in advance!

HaarigerHarald, 2020-09-27:
I'd guess someone introduced an  OMX_RENDER_TRANSITION in one of the openmaxil libs. Can you try to rename the OMX_RENDER_TRANSITION in omx_render.h to something else and do a:

    make clean all

isThatHim, 2020-10-02:
Hello!
I renamed the struct to OMX_RENDER_TRANSITION_RENAME in omx_render.h.

Now i am getting the following compilation error:

> In file included from omxiv.c:14:
omx_render.h:87:31: error: field 'transition' has incomplete type
   87 |  struct OMX_RENDER_TRANSITION transition;
      |                               ^~~~~~~~~~
make: *** [Makefile.include:12: omxiv.o] Error 1

isThatHim, 2020-10-02:
Hello, 
there was a "typedef" missing in omx_render.h in.

If i add the "typedef", it compiles without problems.

`In file included from omxiv.c:14:
omx_render.h:87:31: error: field 'transition' has incomplete type
   87 |  struct OMX_RENDER_TRANSITION transition;
      |                               ^~~~~~~~~~
make: *** [Makefile.include:12: omxiv.o] Error 1
`

isThatHim, 2020-10-02:
Hello,
i could solve the issue by adding "typdef" infront of the `strut OMX_RENDER_TRANSITION` in omx_render.h.

Everything compiled flawlessly now.

Ruffio, 2020-10-06:
@isThatHim hos about s PR?

PR #35 (2021-01-13): multiple definition of OMX_RENDER_TRANSITION - fixes build failure with gcc-10
Fixes a build failure with gcc-10. Resolve #33.

```
cc -o omxiv.bin omxiv.o omx_image.o omx_render.o soft_image.o ./libnsbmp/libnsbmp.o ./libnsgif/libnsgif.o -lilclient -ljpeg -lpng -lrt -ldl -Wl,--gc-sections -s -L/opt/vc/lib/ -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
/usr/bin/ld: omx_render.o:/home/pi/RetroPie-Setup/tmp/build/omxiv/omx_render.h:71: multiple definition of `OMX_RENDER_TRANSITION'; omxiv.o:/home/pi/RetroPie-Setup/tmp/build/omxiv/omx_render.h:71: first defined here
collect2: error: ld [... cut]
closed 2021-01-13 by HaarigerHarald