HaarigerHarald/omxiv #37: make: *** [Makefile:13: omxiv.bin] Error 1

muhammadandykmaulana, 2021-02-17:
Hai everyone...

After cloned from this repo (`https://github.com/HaarigerHarald/omxiv.git`) to Raspberry Pi
and `make ilclient` successfully, and then tried to run `make`.

Unfortunately the command fails with:
```
cc -o omxiv.bin omxiv.o omx_image.o omx_render.o soft_image.o ./libnsbmp/libnsbmp.o ./libnsgif/libnsgif.o -lilclient -ljpeg -lpng -lrt -ldl -Wl,--gc-sections -s -L/opt/vc/lib/ -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
/usr/bin/ld: skipping incompatible /opt/vc/lib//libopenmaxil.so when searching for -lopenmaxil
/usr/bin/ld: cannot find -lopenmaxil
/usr/bin/ld: skipping incompatible /opt/vc/lib//libbcm_host.so when searching for -lbcm_host
/usr/bin/ld: cannot find -lbcm_host
/usr/bin/ld: skipping incompatible /opt/vc/lib//libvcos.so when searching for -lvcos
/usr/bin/ld: cannot find -lvcos
collect2: error: ld returned 1 exit status
make: *** [Makefile:13: omxiv.bin] Error 1
```
So, i make change on script `Makefile.include`:
LDFLAGS+=-L$(SDKSTAGE)/opt/vc/lib/ -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
to
LDFLAGS+=-L$(SDKSTAGE)/opt/vc/lib -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient

```
cc -o omxiv.bin omxiv.o omx_image.o omx_render.o soft_image.o ./libnsbmp/libnsbmp.o ./libnsgif/libnsgif.o -lilclient -ljpeg -lpng -lrt -ldl -Wl,--gc-sections -s -L/opt/vc/lib -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
/usr/bin/ld: skipping incompatible /opt/vc/lib/libopenmaxil.so when searching for -lopenmaxil
/usr/bin/ld: cannot find -lopenmaxil
/usr/bin/ld: skipping incompatible /opt/vc/lib/libbcm_host.so when searching for -lbcm_host
/usr/bin/ld: cannot find -lbcm_host
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcos.so when searching for -lvcos
/usr/bin/ld: cannot find -lvcos
collect2: error: ld returned 1 exit status
make: *** [Makefile:13: omxiv.bin] Error 1
```
After searching, I found `https://stackoverflow.com/questions/16710047/usr-bin-ld-cannot-find-lnameofthelibrary,` but haven't been able to solve the problem. How do I solve this? Thanks

HaarigerHarald, 2021-02-17:
Are you using a x64 OS?

muhammadandykmaulana, 2021-02-17:
Yes Sir, i use amd64 x86_64. I installed Raspi on VM Ware Workstation for testing only.

HaarigerHarald, 2021-02-17:
If the guest system in your VM is a 64 bit system as well it won't compile because openmaxil is only supported on 32bit platforms. Nevertheless, executing this within a VM won't work anyways, since the required HW is missing.

muhammadandykmaulana, 2021-02-18:
Oke, thanks. Can i implement on Raspberry Pi 4? AFAIK, Raspi 4 has SoC: Broadcom BCM2711B0 quad-core A72 (ARMv8-A) 64-bit @ 1.5GHz.

muhammadandykmaulana, 2021-02-24:
Note: only running on Raspi hardware

closed 2021-02-24 by muhammadandykmaulana