HaarigerHarald/omxiv #38: Error following make install

Gene-W, 2021-03-29:
I was having the same issue as #17 and after following those instructions `sudo apt-get update && sudo apt-get -y upgrade && sudo apt-get -y dist-upgrade && sudo rpi-update && sudo reboot` I was able to install omxiv.

Except now at the steps
```
make ilclient 
make sudo 
make install
```

I get the following:

```
make
cc -o omxiv.bin omxiv.o omx_image.o omx_render.o soft_image.o ./libnsbmp/libnsbmp.o ./libnsgif/libnsgif.o -lilclient -ljpeg -lpng -lrt -ldl -Wl,--gc-sections -s -L/opt/vc/lib/ -lopenmaxil -lbcm_host -lvcos -lpthread -L./libs/ilclient
/usr/bin/ld: skipping incompatible /opt/vc/lib//libopenmaxil.so when searching for -lopenmaxil
/usr/bin/ld: cannot find -lopenmaxil
/usr/bin/ld: skipping incompatible /opt/vc/lib//libbcm_host.so when searching for -lbcm_host
/usr/bin/ld: cannot find -lbcm_host
/usr/bin/ld: skipping incompatible /opt/vc/lib//libvcos.so when searching for -lvcos
/usr/bin/ld: cannot find -lvcos
collect2: error: ld returned 1 exit status
make: *** [Makefile:13: omxiv.bin] Error 1
```

I do not know what to try next. This is on a Pi 4 with the 64bit Raspberry Pi OS.

HaarigerHarald, 2021-03-30:
Hi,
yes omxil is apparently deprecated and the libraries haven't and likely won't be ported to arm64. That is why they are missing. I should probably add this info to the README.

Gene-W, 2021-03-30:
> Hi,
> yes omxil is apparently deprecated and the libraries haven't and likely won't be ported to arm64. That is why they are missing. I should probably add this info to the README.

After some time to think about, this is the feeling I had too. Is it still working with 32bit Pi then? I'd love to figure this one out. :)

HaarigerHarald, 2021-03-31:
Yes, if you do install a 32bit OS it should work just fine.

Gene-W, 2021-03-31:
Excellent, thank you for your work on this project and for your help.

closed 2021-03-31 by Gene-W