HaarigerHarald/omxiv #40: Make ilclient error

Joso997, 2021-11-23:
I can no longer install libpng12-dev and are getting:
mkdir -p libs
cp -ru /opt/vc/src/hello_pi/libs/ilclient libs
cp: cannot stat '/opt/vc/src/hello_pi/libs/ilclient': No such file or directory
make: *** [Makefile:25: ilclient] Error 1

ThijmenGThN, 2021-12-01:
Experiencing the same thing, this project seems outdated anyways though.

trimrd, 2021-12-02:
You can achieve the same with OMCV and android apps Screen Stream to mirror phones.
Yatse for everything else. Also, you can stream directly from youtube.

ThijmenGThN, 2021-12-02:
That's indeed true, I've used OSMC before as well, another alternative for Raspberry Pi could be [https://hub.balena.io/gh_rahul_thakoor/balena-rpiplay](url). Lightweight and worth a try.

turzobose, 2022-03-02:
I got the same error. I resolved it by making the iclient inside the omxiv directory. These are the commands:
```
cd omxiv/
make ilclient
make -j4
sudo make install
```
That solved the error for me, and I could stream images and audio from my android device using the Raspicast app.

brunoCarle, 2022-06-20:
@turzobose I got the same error. Unfortunately your workaround doesn't work :

`make ilclient
mkdir -p libs
cp -ru /opt/vc/src/hello_pi/libs/ilclient libs
cp: cannot stat '/opt/vc/src/hello_pi/libs/ilclient': No such file or directory
make: *** [Makefile:25: ilclient] Error 1`

From what I'm reading around, ilclient isn't on the latest images of the Rpi OS ?

For the record, I had to install libjpeg9-dev and libpng-dev instead of libjpeg8-dev and libpng12-dev

kkboy20, 2023-01-27:
ilclient does not work anymore in the latest image. How to install ilclient manually?